MediaTek-Labs/mt3620_m4_software #9: Is OS_HAL recommended?

matsujirushi, 2020-02-11:
Hi,
All sample code use OS_HAL API.
But OS_HAL looks like sample code, not a library.

* Not described in API reference manual. ([here](https://support.mediatek.com/AzureSphere/mt3620/M4_API_Reference_Manual/))
* OS_HAL inc/src are under the Sample_Code directory.

Is OS_HAL recommended or not?
If yes, I want to move out of Sample_Code directory.

LawranceLiu, 2020-02-11:
Hi, 

You are right, the OS_HAL API looks like "sample code" instead of "driver or library".
The purpose of the OS_HAL layer is to simplify the real-time app implementation, while the MHAL layer is to provide the lower-level access to all MT3620 peripherals.

With OS_HAL, App could access external peripherals with simple API.
Without OS_HAL, App could invoke MHAL API directly, however, the App implementation would become more complex.

The OS_HAL API description could be found in the header files: https://github.com/MediaTek-Labs/mt3620_m4_software/tree/master/MT3620_M4_Sample_Code/FreeRTOS/OS_HAL/inc)

MediaTek is open on the usage of OS_HAL.
If OS_HAL API could meet the product requirement, it's suggested to use OS_HAL.
If OS_HAL API could NOT meet the product requirement, customers could use MHAL API directly.

matsujirushi, 2020-02-12:
Thank you for reply.

I think it is better to move OS_HAL to a higher directory.
Allows developers to choose between with OS_HAL and without OS_HAL.

matsujirushi, 2020-02-19:
@LawranceLiu  What is the situation?

LawranceLiu, 2020-02-20:
Hi, 
I'm sorry that I don't catch your point.
What's the problem if we don't move OS_HAL to a higher directory when APP invokes MHAL API directly? 
By the way, which folder would you suggest to move the OS_HAL to?

matsujirushi, 2020-02-20:
Hi,

> What's the problem if we don't move OS_HAL to a higher directory when APP invokes MHAL API directly?

No problem. When APP is calling MHAL, the OS_HAL directory has no effect.

> By the way, which folder would you suggest to move the OS_HAL to?

I think inside MT3620_M4_Driver.

LawranceLiu, 2020-02-22:
Hi Matsuoka-san, 

Thanks for your always valuable suggestion.

Since OS_HAL is "OS dependent" and we would like to keep the MT3620_M4_Driver folder as "OS independent", we think it would be better to keep OS_HAL in the current folder.

And you are welcome to let us know if you have any further suggestions.

matsujirushi, 2020-02-22:
Hi,
That makes sense.

I think OS_HAL is misleading under MT3620_m4_sample_code.
Create MT3620_M4_OS_DEPENDENT directory at the root and put each OS_HAL there?

* /MT3620_M4_OS_DEPENDENT/BareMetal/OS_HAL
* /MT3620_M4_OS_DEPENDENT/FreeRTOS/OS_HAL

LawranceLiu, 2020-02-26:
Hi Matsuoka-san,
Your slide shows the current SW architecture very well: 
https://www.slideshare.net/TakashiMatsuoka2/azure-sphere-227128702/15
All source codes in /MT3620_M4_Sample_Code/ directory are "sample code".
OS_HAL is a kind of sample code, which demonstrates how to use MHAL API, users could invoke the OS_HAL API to simplify the Application source code.
This is the reason we put OS_HAL/ into /MT3620_M4_Sample_Code/.

Since creating /MT3620_M4_OS_DEPENDENT/ introduces a more complicated structure in the root directory, currently, I would suggest not to create this directory.

Could you please help me reconfirm your concern as the following :
    1. The users of this SW package could not easily understand what is OS_HAL/? 
    2. Lack of document to describe the SW layers?
    3. The term "OS_HAL" is misleading and would be better to be renamed as "Middleware"? 
    4. Or does there anything else? 
Please help me figure our your concern correctly, and we could try to figure out the best solution together.

Thank You~

LawranceLiu, 2020-04-13:
Hi Matsuoka-san, 
Please reopen if you have any further question.

closed 2020-04-13 by LawranceLiu