MediaTek-Labs/mt3620_m4_software #13: I2C DMA Problems

brandonhanner, 2020-04-02:
Hey guys, it's me again.

I'm now trying to use I2C to communicate with a peripheral that has transfers in excess of 35 bytes.

The problem I am seeing is when I use the mtk_os_hal_write function, **and** when my input is larger than the FIFO_MAX_LEN of 8, the driver tries to use dma but results in all '0's being sent out onto the bus. When I use fifo mode (ie under 8 bytes) everything works great. What's weird is it seems like the function returns just fine, its just that the wrong bytes are being sent out of the interface. I have already tried splitting my data into multiples of 8 but the slave device wont not accept data that way, although i am successfully able to do the reads in multiples of 8 with no issues. 

I have checked that I have OSAI_ENABLE_DMA in the compile definitions and im using release_200115 which has no changes to the i2c files from the release that just came out. I saw that mhal osai changed so I copied in those changes but no dice.

LawranceLiu, 2020-04-06:
Hi Brandon, 
Thanks for your feedback.
Since the current latest version is release_200320, and the I2C sample code shows the I2C loopback test for length 1B~64B works fine on Seeed board.
I'm wondering if this issue still happen with release_200320 ? 

By the way, do you use Seeed board or AVNET board ? Which ISU did you use ? (ISU0 or ISU1 or ...)
One potential issue I noticed is : 
    On Seeed board, the I2C loopback test result of 1B~64B is OK.
    On AVENT board, the I2C loopback test result of 1B~9B  is OK. but if the length is larger than 9B, the test result becomes unstable. 
    There are two approach to make the test result on AVNET board more stable:
        1. Add pull-high circuit to the I2C bus.
        2. Swich the ISU role, Configure ISU2 be I2C master, and ISU1 as I2C Slave.
If you are using AVNET board and ISU1 as I2C master, maybe you could try to use ISU2.

LawranceLiu, 2020-04-07:
Hi Brandon,
One more reminding for you:
Please don't declare the I2C Tx/Rx buffer as global array.
Please allocate the I2C Tx/Rx buffer using pvPortMalloc( ) or declare it as local array.
The background reason is the DMA  is only supported for M4 SYSRAM, not M4 TCM.
If you declare the buffer as global array, it would be placed in M4 TCM.

brandonhanner, 2020-04-07:
Hey Lawrance, 

I am using I2C Master on ISU2 on the AVNET board. I know it has the lsm6dso on the same bus but I believe it to not have any effect on the other transfers. 

I was able to dig into the drivers yesterday and modify them to handle fifo transfers in excess of 8 bytes (by constantly reading the FIFO_FULL flag and filling the FIFO with respect to that until the transfer is complete) and I have no trouble sending 35bytes of data back to back on the bus. 

Looking at the logic analyzer with a transfer under 8 bytes everything goes according to plan, but as soon as dma is used, the i2c master successfully sets up a transfer to the correct address (including ack from slave) but sends 35 bytes of all 0's. It feels like the operation of DMA transferring the bytes to the i2c peripheral is failing or perhaps the placement of the data into dma region itself is failing. I'm not exactly sure how to verify what's going on in the DMA region. 

I tried using the latest release just to double check before making this reply but i am seeing the same behavior.

I included the sample I used with the latest release that I modified to make a larger transfer.
*NOTE: you have to pause the debugger and change "f" of main to 1 in order for the program to proceed, or remove that while loop alltogether. This is in place to make sure debugger is in sync with sphere while debugging.

```
/*
 * (C) 2005-2019 MediaTek Inc. All rights reserved.
 *
 * Copyright Statement:
 *
 * This MT3620 driver software/firmware and related documentation
 * ("MediaTek Software") are protected under relevant copyright laws.
 * The information contained herein is confidential and proprietary to
 * MediaTek Inc. ("MediaTek"). You may only use, reproduce, modify, or
 * distribute (as applicable) MediaTek Software if you have agreed to and been
 * bound by this Statement and the applicable license agreement with MediaTek
 * ("License Agreement") and been granted explicit permission to do so within
 * the License Agreement ("Permitted User"). If you are not a Permitted User,
 * please cease any access or use of MediaTek Software immediately.
 *
 * BY OPENING THIS FILE, RECEIVER HEREBY UNEQUIVOCALLY ACKNOWLEDGES AND AGREES
 * THAT MEDIATEK SOFTWARE RECEIVED FROM MEDIATEK AND/OR ITS REPRESENTATIVES ARE
 * PROVIDED TO RECEIVER ON AN "AS-IS" BASIS ONLY. MEDIATEK EXPRESSLY DISCLAIMS
 * ANY AND ALL WARRANTIES, EXPRESS OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE
 * IMPLIED WARRANTIES OF MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE OR
 * NONINFRINGEMENT. NEITHER DOES MEDIATEK PROVIDE ANY WARRANTY WHATSOEVER WITH
 * RESPECT TO THE SOFTWARE OF ANY THIRD PARTY WHICH MAY BE USED BY,
 * INCORPORATED IN, OR SUPPLIED WITH MEDIATEK SOFTWARE, AND RECEIVER AGREES TO
 * LOOK ONLY TO SUCH THIRD PARTY FOR ANY WARRANTY CLAIM RELATING THERETO.
 * RECEIVER EXPRESSLY ACKNOWLEDGES THAT IT IS RECEIVER'S SOLE RESPONSIBILITY TO
 * OBTAIN FROM ANY THIRD PARTY ALL PROPER LICENSES CONTAINED IN MEDIATEK
 * SOFTWARE. MEDIATEK SHALL ALSO NOT BE RESPONSIBLE FOR ANY MEDIATEK SOFTWARE
 * RELEASES MADE TO RECEIVER'S SPECIFICATION OR TO CONFORM TO A PARTICULAR
 * STANDARD OR OPEN FORUM. RECEIVER'S SOLE AND EXCLUSIVE REMEDY AND MEDIATEK'S
 * ENTIRE AND CUMULATIVE LIABILITY WITH RESPECT TO MEDIATEK SOFTWARE RELEASED
 * HEREUNDER WILL BE ANY SOFTWARE LICENSE FEES OR SERVICE CHARGE PAID BY
 * RECEIVER TO MEDIATEK DURING THE PRECEDING TWELVE (12) MONTHS FOR SUCH
 * MEDIATEK SOFTWARE AT ISSUE.
 */

#include "FreeRTOS.h"
#include "task.h"
#include "printf.h"
#include "mt3620.h"

#include "os_hal_uart.h"
#include "os_hal_i2c.h"

/*
 *Additional Note:
 *This sample code uses ISU1 as I2C Master and ISU2 as I2C Slave for the I2C
 *loopback test. ISU1(SDA/SDL) should be connected to ISU2(SDA/SDL).
 *Note, This sample code works only on Seeed development board, NOT on AVNET
 *development board. Since ISU2 of AVNET development board has an additional
 *sensor connected, if run this sample code on AVNET development board, some I2C
 *transport error might be happening. One possible work around for AVNET
 * development board is to change I2C_MAX_LEN from 64 to 8.
*/

/******************************************************************************/
/* Configurations */
[... 79 lines cut ...]
_Noreturn void RTCoreMain(void)
{
	volatile bool f = false;
	while (!f) {
		// empty
	}
	/* Setup Vector Table */
	NVIC_SetupVectorTable();

	/* Init UART */
	mtk_os_hal_uart_ctlr_init(uart_port_num);
	printf("\nFreeRTOS I2C Demo\n");

	/* Init I2C Master/Slave */
	mtk_os_hal_i2c_ctrl_init(i2c_master_port_num);

	/* Create I2C Master/Slave Task */
	xTaskCreate(i2c_master_task, "I2C Master Task", APP_STACK_SIZE_BYTES,
		NULL, 4, NULL);

	vTaskStartScheduler();
	for (;;)
		__asm__("wfi");
}
```

LawranceLiu, 2020-04-07:
Hi Brandon, 
Thanks for your sharing. 
I believe this issue could be fixed by:
Change from : 
		mtk_os_hal_i2c_write(i2c_master_port_num,
			0x60, randombytes, 35);
To :
                memcpy(write_buf, randombytes, 35);
		mtk_os_hal_i2c_write(i2c_master_port_num,
			0x60, write_buf, 35);
The key point is described in my previous post: The DMA is only supported for M4 SYSRAM, not M4 TCM. When you try to do DMA, you have to make sure the buffer address is get from pvPortMalloc( )  or local array. We will update our DMA driver later to print some error log when the DMA buffer address is in TCM.

brandonhanner, 2020-04-07:
Ahh ok. Sorry I did not see your post just before my response. 

Thanks for the clarity!

brandonhanner, 2020-04-09:
Just wanted to confirm the memory situation was the issue. After declaring the variable locally everything works. 

Thanks again!

closed 2020-04-09 by brandonhanner