MediaTek-Labs/mt3620_m4_software #16: How to use I2S interface?

matsujirushi, 2020-05-01:
I changed MHAL_I2S_TYPE_INTERNAL_LOOPBACK_MODE to  MHAL_I2S_TYPE_EXTERNAL_MODE at MT3620_RTApp_FreeRTOS_I2S.

[image]

But no signals without MCLK. (TX, FS and SCLK).

[image]

LawranceLiu, 2020-05-01:
You mean there is signal on MCLK, but no signal on TX/FS/SCLK, right?
SCLK/FS should be generated by I2S Master, and TX would be generated by MT3620 according to the SCLK signal.

matsujirushi, 2020-05-02:
Hi Lawrance-san,
Thanks for your information.
Sorry. I was wrong.
I thought MT3620 supported I2S master.

"Two I2S interfaces supporting slave and TDM slave modes"
[MT3620 Datasheet](https://d86o2zu8ugzlg.cloudfront.net/mediatek-craft/documents/mt3620/MT3620-Datasheet-v1.4.pdf)
[MT3620 M4 API Reference Manual](https://support.mediatek.com/AzureSphere/mt3620/M4_API_Reference_Manual/)

closed 2020-05-02 by matsujirushi