MediaTek-Labs/mt3620_m4_software #18: Periodic ADC Sample

matsujirushi, 2020-05-05:
Hi,
This driver has ADC API for periodic. [mtk_os_hal_adc_period_get_data()](https://support.mediatek.com/AzureSphere/mt3620/M4_API_Reference_Manual/group__os__hal__adc__function.html#ga94333a9f7e23ffd4ffd89aea6a3ff7fe)
I don't know how to call it.

Could you tell me sample code, code snippet or hint?

matsujirushi, 2020-05-06:
I want to convert Analog-to-Digital continuously about 10KHz.

LawranceLiu, 2020-05-15:
We are now working on this, and will get back to you as soon as possible.

LawranceLiu, 2020-08-28:
ADC period mode driver API and RTApp sample code are updated in [release_200828](https://github.com/MediaTek-Labs/mt3620_m4_software/commit/88b47b1e9cb26307fda02e6643846f0ca60ca591)

LawranceLiu, 2020-08-31:
Close this issue due to the solution is included in release_200828. Please reopen if any further issues found.

closed 2020-08-31 by LawranceLiu