MediaTek-Labs/mt3620_m4_software #19: [GPT3] accuracy of period

matsujirushi, 2020-05-09:
When 250usec.(4KHz) is set to GPT3 period, little slow. (264usec.)

**Code:**
https://github.com/matsujirushi/mt3620_m4_software/tree/matsujirushi-issue-1/MT3620_M4_Sample_Code/FreeRTOS/MT3620_RTApp_FreeRTOS_GPT
https://github.com/matsujirushi/mt3620_m4_software/blob/matsujirushi-issue-1/MT3620_M4_Sample_Code/FreeRTOS/MT3620_RTApp_FreeRTOS_GPT/main.c#L57

**Result:**
[image]

Is it design?

LawranceLiu, 2020-05-15:
Hi Matsuoka-san, 

You could try to figure out it's caused by HW or SW with the following approach: Restore the gpt3_interval to 5,000,000 (5 seconds) and check the console log.

In my test environment, the log shows:
    TimerHandlerGpt3 (SysTick=5002)
    TimerHandlerGpt3 (SysTick=9995)
    TimerHandlerGpt3 (SysTick=14987)
    TimerHandlerGpt3 (SysTick=19979)
    TimerHandlerGpt3 (SysTick=24971)

The delta time between each Gpt3 timeout is about 4992ms in my test environment.
It means there is a 8ms deviation for every 5000ms, which means there would be a 0.4us deviation for every 250us. 

If you see similar result in your evnironment (8ms deviation for every 5000ms) but the measured Gpt3 timeout interval for 250us is 264us, this might be caused by SW issue. 

Did you toggle GPIO when Gpt3 timeout? To speed up, you could try to invoke the lowest layer API osai_writel() instead of OS_HAL API mtk_os_hal_gpio_set_output().

matsujirushi, 2020-05-15:
Hi Lawrance-san,
Thanks for your advice.
Again, I checked.

https://github.com/matsujirushi/mt3620_m4_software/tree/matsujirushi-issue-19/MT3620_M4_Sample_Code/FreeRTOS/MT3620_RTApp_FreeRTOS_GPT

**Code:**
```cpp
static const uint32_t gpt_timer_3_interval = 250;
```

**Result:**
264usec.
[image]

**Code:**
```cpp
static const uint32_t gpt_timer_3_interval = 5000000;
```

**Result:**
4.999939sec.
[image]

LawranceLiu, 2020-05-18:
Hi Matsuoka-san, 

According to your result, it seems the timer HW accuracy is good.
There is only 61us deviation per 5,000,000us, and it's expected there is only 0.003us deviation per 250us.

I doubt the 14us (264us-250us) is caused by SW.
Ex, additional latency of the interrupt handler, additional latency of the OS_HAL GPIO APIs invocation.

Since there might be different deviations on different boards or different test environments.
If you would like to have the precise 250us timer, I would suggest you set the timer to a value smaller than 250.

matsujirushi, 2020-06-11:
Hi Laurance-san,
Sorry for delay.
I changed the intervals and took some measurements.
The GPIO output was about 14 usec. long for the interval I set.
You're right, I think the software overhead is adding up because GPT3 can't auto reset.

Thanks for your advice.

[image]

closed 2020-06-11 by matsujirushi