MediaTek-Labs/mt3620_m4_software #21: warning : command line option '-std=c11' is valid for C/ObjC but not for C++

matsujirushi, 2020-05-22:
When compiling MT3620_RTApp_FreeRTOS_CPP, warning occurs.

**warning : command line option '-std=c11' is valid for C/ObjC but not for C++**

```
  [1/51] C:\PROGRA~2\MICROS~2\2019\PROFES~1\Linux\gcc_arm\bin\AR10B2~1.EXE  -DOSAI_ENABLE_DMA -DOSAI_FREERTOS -I../../FreeRTOS_RTcore_CPP -I../../PUBLIC -I../../. -I../../../../OS_HAL/inc -IC:/AzureSphere/azure-sphere-thief-detector/mt3620_m4_software/MT3620_M4_Driver/../MT3620_M4_BSP/printf -IC:/AzureSphere/azure-sphere-thief-detector/mt3620_m4_software/MT3620_M4_Driver/./MHAL/inc -IC:/AzureSphere/azure-sphere-thief-detector/mt3620_m4_software/MT3620_M4_Driver/./HDL/inc -IC:/AzureSphere/azure-sphere-thief-detector/mt3620_m4_software/MT3620_M4_Driver/../MT3620_M4_Sample_Code/OS_HAL/inc -IC:/AzureSphere/azure-sphere-thief-detector/mt3620_m4_software/MT3620_M4_BSP/./CMSIS/include -IC:/AzureSphere/azure-sphere-thief-detector/mt3620_m4_software/MT3620_M4_BSP/./mt3620/inc -IC:/AzureSphere/azure-sphere-thief-detector/mt3620_m4_software/MT3620_M4_BSP/FreeRTOS/./portable -IC:/AzureSphere/azure-sphere-thief-detector/mt3620_m4_software/MT3620_M4_BSP/FreeRTOS/./include -isystem "C:/Program Files (x86)/Microsoft Visual Studio/2019/Professional/Linux/gcc_arm/lib/gcc/arm-none-eabi/8.2.1/include" -isystem "C:/Program Files (x86)/Microsoft Visual Studio/2019/Professional/Linux/gcc_arm/arm-none-eabi/include" -std=gnu++14 -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -Wall -g   -std=c11 -Wall -g2 -gdwarf-2 -O0 -MD -MT CMakeFiles/FreeRTOS_RTcore_CPP.dir/main.cpp.obj -MF CMakeFiles\FreeRTOS_RTcore_CPP.dir\main.cpp.obj.d -o CMakeFiles/FreeRTOS_RTcore_CPP.dir/main.cpp.obj -c ../../main.cpp
C:\AzureSphere\azure-sphere-thief-detector\mt3620_m4_software\MT3620_M4_Sample_Code\FreeRTOS\MT3620_RTApp_FreeRTOS_CPP\out\ARM-Debug\cc1plus.exe : warning : command line option '-std=c11' is valid for C/ObjC but not for C++
```

LawranceLiu, 2020-05-22:
Thanks for the reminding.
It's fixed.

matsujirushi, 2020-05-25:
Thanks!
https://github.com/MediaTek-Labs/mt3620_m4_software/commit/bd360eb16cef96ae0317c513608fc4450c4a88fe

closed 2020-05-25 by matsujirushi