MediaTek-Labs/mt3620_m4_software #22: Constructor of static variable is not called

matsujirushi, 2020-06-12:
Constructor of static variable is not called.
Don't you know how to fix it?

[image]

LawranceLiu, 2020-06-15:
Declare class objects as static/global variables would lead to constructor invocation timing uncontrollable. In this case, the constructor seems not invoked.
You could try this way: 

class DigitalOut
{
private:
	os_hal_gpio_pin _Pin;

public:
	DigitalOut(os_hal_gpio_pin pin);
	DigitalOut(){};
	void Write(int value);
};

DigitalOut::DigitalOut(os_hal_gpio_pin pin)
{
	_Pin = pin;
	mtk_os_hal_gpio_request(_Pin);
	mtk_os_hal_gpio_set_direction(_Pin, OS_HAL_GPIO_DIR_OUTPUT);
}

void DigitalOut::Write(int value)
{
	mtk_os_hal_gpio_set_output(_Pin, value ? OS_HAL_GPIO_DATA_HIGH : OS_HAL_GPIO_DATA_LOW);
}

static DigitalOut led;

static void blink_task(void* pParameters)
{
	led = DigitalOut(gpio_led_green);
        //....
}

matsujirushi, 2020-06-16:
Hi Lawrance-san,

In this case, this code looks fine.
However, I think the lack of a constructor of static variable call should be addressed in the development environment, not in the code.

LawranceLiu, 2020-06-23:
Hi Matsuoka-san, 
Thank you very much for the solution!

closed 2020-07-21 by LawranceLiu