MediaTek-Labs/mt3620_m4_software #24: [Question] Ethernet on M4

Visit-Learn, 2020-07-21:
Hi,

I want to know there has sample code for Ethernet IC (such as ENC28J60) to do some simple internet 
application for us to reference?

Thank you so much.

kevinwh-chou, 2020-07-21:
Hi, 
 
As the network function is managed by CA7 core instead of CM4 core, you need to refer to the following link from Microsoft Azure Sphere website for the steps to enable Ethernet function with ENC28J60.
https://docs.microsoft.com/en-us/azure-sphere/network/connect-ethernet

And there's also a bunch of samples for Azure Sphere (on CA7 core), like the following link shows an example how you can connect an Azure Sphere device to a private network and use network services.
https://github.com/Azure/azure-sphere-samples/tree/master/Samples/PrivateNetworkServices

Visit-Learn, 2020-07-21:
Hi,

Thank you, but as far as we know, the CA7 only supports one Ethernet port, but we want to use two ENC28J60 for our application. One we could use [](https://docs.microsoft.com/en-us/azure-sphere/network/connect-ethernet), another
we only could select CM4 to implement it (which could not have Azure sphere ability, we only want to use this Ethernet to connect other sensor device). Do you have any suggestion on MT3620 to support two SPI Ethernet?

Thank you so much.

kevinwh-chou, 2020-07-22:
Hi,

CM4 is supposed to be able to control, receive and transmit data with ENC28J60 via SPI, however it's expected that all the network data should go through CA7 majorly out of security concerns, so there is no sample code available on CM4 to connect with ENC28J60. Besides, the TCP/IP protocol stack is implemented on CA7, therefore it would be more suitable and easier to have you application run on it. 

May I know the reason to use two Ethernet ports, can the sensor be connected to a Ethernet switch first then to MT3620's Ethernet port on CA7? Or one port is needed for public network and another for private network where the sensor device locates? And what's the network protocol operating between the sensor device and MT3620?

Visit-Learn, 2020-07-28:
Hi

> CM4 is supposed to be able to control, receive and transmit data with ENC28J60 via SPI, however it's expected that all the network data should go through CA7 majorly out of security concerns, so there is no sample code available on CM4 to connect with ENC28J60. Besides, the TCP/IP protocol stack is only implemented on CA7, therefore it would be more suitable and easier to have you application run on it.
> 
**<Reply>** Understand, thanks.

> May I know the reason to use two Ethernet ports, can the sensor be connected to a Ethernet switch first then to MT3620's Ethernet port on CA7? Or one port is needed for public network and another [...]

**<Reply>** Actually, we want to use M4 for Ethernet to build up simple protocol (such as uIP or LwIP) for internal network with different control or sensors and A7 for fully network support~ 

I will close this question this week if no other related asking, thank you so much.

closed 2020-08-03 by Visit-Learn