MediaTek-Labs/mt3620_m4_software #26: [Question] GPIO Interrupt on M4

Visit-Learn, 2020-07-28:
Hi,

I check your sample code for GPIO (MT3620_RTApp_FreeRTOS_GPIO), and I found that is using polling method to know the button is pressed or not; but I think that using GPIO interrupt is more suitable for MCU limited resource. Could I know there has material or sample code to describe it? Thank you so much.

kevinwh-chou, 2020-07-28:
Hi,

You're right, but for the moment external interrupt function is not supported in Azure Sphere OS firewall configuration, so in the example it uses the polling method instead of receiving an external interrupt, as the latter is not available yet.

Visit-Learn, 2020-07-28:
Hi,

Got it, thank you so much.

closed 2020-07-28 by Visit-Learn

kevinwh-chou, 2020-07-28:
You're welcome!
Besides that, the purpose of that GPIO sample code is to demonstrate the basic GPIO functions, like reading a state or setting an output state, so it reads a state from the button and outputs a state at LED just as the example.