MediaTek-Labs/mt3620_m4_software #30: FreeRTOS sample MT3620_RTApp_FreeRTOS_SPIM does not build

JakeSays, 2020-08-11:
The sample MT3620_RTApp_FreeRTOS_SPIM does not build due to improper use of fpu instructions.

Error: selected FPU does not support instruction -- `vstmdbeq r0!,{s16-s31}'

The issue is in mt3620_m4_software/MT3620_M4_BSP/FreeRTOS/portable/port.c, lines 445 and 467.

kevinwh-chou, 2020-08-12:
Hi,

Have you tried below? If no, please give it a try and let me know if it works for you or not.

To build MediaTek FreeRTOS sample code, the GCC compiler flag should be modified to use FPU instructions. 
Copy “AzureSphereRTCoreToolchainVFP.cmake” from MediaTek sample code folder to Azure Sphere SDK install folder.
(by default C:\Program Files (x86)\Microsoft Azure Sphere SDK\CMakeFiles)

kevinwh-chou, 2020-08-13:
Hi, 

BTW, if you use the latest sample code / driver from the Github, it doesn't necessarily require to copy that AzureSphereRTCoreToolchainMTK.cmake to the Azure Sphere SDK folder as the latest codebase has been updated to use the file from the same sample code folder. 

So I would like to confirm if you use the latest software / driver, or have you done any modifications to the M4 BSP part? 
According to the internal testing result, MT3620_RTApp_FreeRTOS_SPIM is supposed to work fine, and there's no need to make changes.

JakeSays, 2020-08-13:
I pulled latest from the repo, opened the project in visual studio code, and built. I didn't make any changes.

kevinwh-chou, 2020-08-13:
Hi, 

Got it, then it's strange, they should work fine. 
What's the Azure Sphere SDK version currently installed? 20.04, 20.07, or other?

JakeSays, 2020-08-19:
@kevinwh-chou I'm on 20.07

closed 2020-12-17 by kevinwh-chou