MediaTek-Labs/mt3620_m4_software #31: [Question] UART Rx interrupt

Visit-Learn, 2020-08-12:
Hi,

I refer MTK FreeRTOS example for UART. And I found that the reception is using polling method. 
And I saw below link without FreeRTOS from CodethinkLabs is implement by Rx interrupt, so could I know
how to use interrupt method to deal with UART receive based on MTK SDK?
[CodethinkLabs UART Example based On Interrupt](https://github.com/CodethinkLabs/mt3620-m4-samples/blob/master/UART_RTApp_MT3620_BareMetal/main.c)

If not, could I know the UART has hardware buffer? Because I saw example the task is every 10 ms to polling, I think
that too slow to deal with 115200 bps, thank you so much.

kevinwh-chou, 2020-08-13:
Hi,

The current UART driver is implemented for polling mode, but not support interrupt mode. To get the data at quicker pace, the interval can be reduced to shorter than 10ms. It's not necessary to use 10ms, if the data is exchanged at higher baud rate. The hardware buffer for Rx is 32byte, so every time the amount of data you read is 32 byte.

You may see some functions seemed to work for interrupt mode, but they're either not in use now (ex. mtk_os_hal_uart_set_irq()) or only used for internal processing in DMA mode (ex. call-back functions, _mtk_os_hal_uart_dma_rx_callback()).

kevinwh-chou, 2020-08-13:
The mtk_os_hal_uart_dma_send_data() function also works based on interrupt, it waits until RX is done then go on to the next step, so another approach is creating another task with mtk_os_hal_uart_dma_send_data() in it to receive and move data.

```
mtk_mhal_uart_start_dma_rx(ctlr);
ret = _mtk_os_hal_uart_wait_for_rx_done(ctlr_rtos, timeout);
```

Just be noted that in the freeRTOS task, a small delay in the while loop is still needed for context switching.

Visit-Learn, 2020-08-25:
Hi,

Understood, thank you so much.

closed 2020-08-25 by Visit-Learn