MediaTek-Labs/mt3620_m4_software #32: MT3620_RTApp_BareMetal_HelloWorld does not build

matsujirushi, 2020-08-30:
When create CMake cache, I got error below.
```
[CMake] CMake Error: The source directory
```

[image]

[log.txt](https://github.com/MediaTek-Labs/mt3620_m4_software/files/5146309/log.txt)

Azure Sphere SDK is 20.07.

kevinwh-chou, 2020-08-31:
Hi, 

The project on the github should be ok, but similar error message seems to pop up in some building environments

 in some other case.

Can you test the following step first to see if it works in your environment or not?
1. Replace the original CMakeLists.txt and CMakeSettings.json in the app folder with the attached ones.
[cmake.zip](https://github.com/MediaTek-Labs/mt3620_m4_software/files/5149378/cmake.zip)

If error still exists, remove the following lines in the updated CMakeSettings.json
```
        {
          "name": "AZURE_SPHERE_DEFAULT_CMAKE_PATH",
          "value": "${env.AzureSphereDefaultSDKDir}CMakeFiles\\"
        }
```

matsujirushi, 2020-08-31:
Hi,
Thank you for reply.

> Replace the original CMakeLists.txt and CMakeSettings.json in the app folder with the attached ones, then generate cmake cache.

[image]

> If the error still exists, remove the following lines in the updated CMakeSettings.json and generate cache again.

[image]

[error2.txt](https://github.com/MediaTek-Labs/mt3620_m4_software/files/5150044/error2.txt)

kevinwh-chou, 2020-09-01:
Hi,

Thanks for the update, then it looks to be a different case. 
This kind of error seems to occur in some build environments after upgrading Azure Sphere SDK to 20.07, and strangely it can't be seen in our environments.

Did you try this code with the previous Azure Sphere SDK 20.04?
And did you also try other sample code (like samples for freeRTOS) but the error still existed?

kevinwh-chou, 2020-09-01:
Hi,

Can you try the following moves again with the attachment to see if it works or not? 
[cmake_20200901.zip](https://github.com/MediaTek-Labs/mt3620_m4_software/files/5155654/cmake_20200901.zip)

1. Replace the original CMakeLists.txt and CMakeSettings.json in the app folder with the attached ones. 
2. Copy the AzureSphereRTCoreToolchainMTK.cmake in the attachment to CMakeFiles folder in Azure Sphere SDK directory (like mine is C:\Program Files (x86)\Microsoft Azure Sphere SDK\CMakeFiles\). 

Thanks!

matsujirushi, 2020-09-01:
Hi @kevinwh-chou ,
It work!

Copy cmake_20200901.zip to project folder.
(No need to copy AzureSphereRTCoreToolchainMTK.cmake.)

If you want, I can test all projects and create PR.

LawranceLiu, 2020-09-01:
Hi Matsuoka-san, 

Thank you very much for your kindly support.
We have figured out some solutions (which are a little bit different for BareMetal and FreeRTOS) and will put to GitHub soon.

This issue seems to happen only on some Windows PC. In the past few days, I could not reproduce this build error problem until today I got a new PC and installed the Visual Studio & Azure Sphere SDK from scratch. This issue seems to be caused by some environment variable configuration changes due to SDK update. We will adjust the environment variable configurations in the CMake files.

LawranceLiu, 2020-09-11:
Hi Matsuoka-san,
This issue is fixed in release_200911.
If you see any further problems, you could reopen this issue.

closed 2020-09-11 by LawranceLiu