MediaTek-Labs/mt3620_m4_software #34: [Question] About CM4F0 and CM4F1

Visit-Learn, 2020-11-04:
Hi,

we refer mailbox example for using two RT core. And we have some questions as following,
1. How could I assign RT core firmware into specific M4 part
    (such that we only have one RT core firmware but we want to assign it into CM4F1)

2. If only one RT core firmware and this firmware does not use any dedicate peripheral (such as IO0_GPIO86)
    How could we know that firmware is side load into CM4F0 or CM4F1

Thank you so much.

kevinwh-chou, 2020-11-04:
Hi,

For your questions, please refers to "8. Determine which core your app is running on" section from [here](https://docs.microsoft.com/zh-tw/azure-sphere/app-development/develop-debug-rt-app)
Basically the RTApp is deployed to the first available real-time core on the device, and it's not able to specify a particular core. Inside the link it shows a way to know which core is currently used.

Visit-Learn, 2020-11-04:
Hi,

Thank you for your information, got it.
Thanks~

closed 2020-11-04 by Visit-Learn