MediaTek-Labs/mt3620_m4_software #35: [Question] IO0_GPIO85 to IO1_GPIO92

Visit-Learn, 2020-11-04:
Hi,

Could we know that IO0_GPIO85 to IO1_GPIO92 can use as GPIO by our application?
Currently, we meet one issue, we use MT3620_RTApp_FreeRTOS_GPIO and we only add IO0_GPIO86 into app_manifest.json, such as following, 
[image]
Then when we side load this package, the error occurs as following,
[image]

Could we know there is document to describe it? Thank you so much.

kevinwh-chou, 2020-11-04:
Hi,

IO0_GPIO85 to IO1_GPIO92 are special cases, they're dedicated to certain cores, so I think it should be like the case of using dedicated UART for certain core that it doesn't require to be claimed in app_manifest. If you do so, I guess VS may throw an error as it's not part of general GPIOs.

Visit-Learn, 2020-11-04:
Hi,

Actually, there has no build code error when we add 86 into manifest. The error happened when we side load firmware.
And we notice those pins are for dedicated usage, but we also see data sheet as below,
[image]

So we have a little confused about it and we just want to double check with you the IO0_GPIO85 to IO1_GPIO92 could only
be used as UART, is it correct? If any misunderstanding, please feel free to correct me.

Thank you so much.

kevinwh-chou, 2020-11-05:
Hi, 

I think what you see makes sense as to my understanding app_manifest affects the sideload step and later operation but not involved in image build/compilation stage. 

From HW perspective, GPIO is indeed one option in the pinmux for that pin, however according to the comment in [this Azure Sphere page](https://docs.microsoft.com/zh-tw/azure-sphere/app-development/develop-debug-rt-app#log-output-from-an-rtapp) for dedicated UART, "it shouldn't be used for purposes other than logging output for an RTApp", I think Microsoft may block the user from using these pins for GPIOs, as a result an error is probably thrown. 

Besides that, some of those dedicated pins are also used for strapping functions, suggest you to use the GPIO pins from the shared peripherals than the core dedicated pins.

Visit-Learn, 2020-11-05:
Hi,

Thank you very much. Got it.

closed 2020-11-05 by Visit-Learn

nishithvpoojary, 2021-02-19:
I have read M4 dedicated UART is intended for logging purpose. What does logging mean?  And What is the difference between the ISU UART and Dedicated UART?

LawranceLiu, 2021-02-19:
[image]
[image]

Each CM4 core has it own UART interface, we call it "Dedicated UART".
The difference between the "ISU UART" and "Dedicated UART" is:
    *. "ISU UART" could be assigned to CA7 or CM4_0 or CM4_1 by configuring the "app_manifest.json".
    *. "Dedicated UART 0" could be only accessed by CM4_0. (You don't need to configure the "app_manifest.json".)
    *. "Dedicated UART 1" could be only accessed by CM4_1. (You don't need to configure the "app_manifest.json".)

Regarding the term "logging", it means "debug log".

nishithvpoojary, 2021-02-19:
How to assign ISU to CM4_0 or CM4_1 ? . I am using ISU0 in app_manifest.json

LawranceLiu, 2021-02-19:
The sample code is available here: 
https://github.com/MediaTek-Labs/mt3620_m4_software/tree/master/MT3620_M4_Sample_Code/FreeRTOS/MT3620_RTApp_FreeRTOS_UART

And the ISU is assigned to CM4 by configuring the app_manifest.json:
https://github.com/MediaTek-Labs/mt3620_m4_software/blob/master/MT3620_M4_Sample_Code/FreeRTOS/MT3620_RTApp_FreeRTOS_UART/app_manifest.json#L8

If you have two RTApps to be downloaded to the MT3630. 
The "first" RTApp you downloaded wouuld be run on CM4_0.
The "second" RTApp you downloaded wouuld be run on CM4_1.

nishithvpoojary, 2021-02-19:
Thank you very much.