MediaTek-Labs/mt3620_m4_software #37: Data Sharing Between Clicks

nishithvpoojary, 2021-02-19:
Is there any code sample to transfer data from Click1 to Click2. I want to transfer the Click1 Rx data to Click2 and send it to PC using FreeRTOS.

LawranceLiu, 2021-02-19:
You could refer to the AVENT Azure Sphere Starter Kit User Guide: 
[image]
The "Rx" pin of the Click1 & Click2 is the same.
What data would you like to transfer?

nishithvpoojary, 2021-02-19:
I connected Tx and Rx of Click1 with jumper wire and sent a printf message. Now I want to transfer that Click 1 Rx data to Click 2 so that I can display it in terminal using click 2. How can I do that?

LawranceLiu, 2021-02-19:
Hi Nishithvpoojary, 
I'm sorry that I don't really catch your point.
The "Click1 Tx/Rx" and "Click2 Tx/Rx" are mapped to the same pins:
    *. Click1_Tx = Click2_Tx
    *. Click1_Rx = Click2_Rx
If you printf to ISU0, the UART Tx single would be detected on both Click1_TX and Click2_Tx.
If you connect Click1_Tx to Click1_Rx, then, it also means Click2_Tx is connected to Click2_Rx.

Could the above answer your question?
Welcome to let me know if anything not clear.

nishithvpoojary, 2021-02-20:
I connected a jumper wire from Click1 Tx to Rx and sent a message "Hello". Now I want to transfer the received message to Click 2 and send from Click 2 Tx to  Terminal to display this message.

kevinwh-chou, 2021-02-22:
Hi,

What's the purpose for loopback between Click1 Tx to Rx, just for testing?
Guess you check whether the data is corrupted or not on the terminal, what about the received data in the loopback in step1?
Check if the result at each step is expected or not.

nishithvpoojary, 2021-02-22:
Yes Kevinwh-chou,
Loopback was for testing purpose. I wanted to check the receiving data. I was able to transfer the data from Click1 to Click2 and the data was not corrupted.

closed 2021-02-22 by nishithvpoojary