MediaTek-Labs/mt3620_m4_software #39: FreeRTOS build errors on Linux development computer

bawilless, 2021-03-01:
I’m trying to build the MT3620_RTApp_FreeRTOS_MBOX/RTAppA example on my Linux development computer.  I get the error shown below.  I can pull the same source and build it on my Windows computer without any errors.  Any idea how to resolve this issue?

```Error: selected FPU does not support instruction -- `vstmdbeq r0!,{s16-s31}'```
```Error: selected FPU does not support instruction -- `vldmiaeq r0!,{s16-s31}'```

The steps to reproduce the issue . . .

- On a Ubuntu development computer
  - Pull the latest source tree
  - Open the MT3620_RTApp_FreeRTOS_MBOX/RTAppA project
  -  At this point I do see a Visual Studio Code warning ```"Warning: Required settings are missing from one or more Azure Sphere projects. Configure these projects now?"```
     - Answer yes to configure the project now and then select the Azure Spehre real-time configuration option  
  - Clean reconfigure the project
  - Clean rebuild the project
- The build fails with the following errors  
  - ```Error: selected FPU does not support instruction -- `vstmdbeq r0!,{s16-s31}'```
  - ```Error: selected FPU does not support instruction -- `vldmiaeq r0!,{s16-s31}'```

Thanks

LawranceLiu, 2021-03-25:
When building the RTApp sample code with Visual Studio Code on Linux, the .vscode/settings.json is generated by VS Code automatically with the following configuration:
"CMAKE_TOOLCHAIN_FILE": "${command:azuresphere.AzureSphereSdkDir}/CMakeFiles/**AzureSphereRTCoreToolchain.cmake**"
It means the Azure Sphere SDK default AzureSphereRTCoreToolchain.cmake is used. 
When building the FreeRTOS RTApp, the FreeRTOS BSP requires the FPU support. and the default cmake file does not turn on the FPU support.

The solution to fix the build error is to modify the Azure Sphere SDK default AzureSphereRTCoreToolchain.cmake (/opt/azurespheresdk/CMakeFiles/AzureSphereRTCoreToolchain.cmake):
set(CMAKE_C_FLAGS_INIT "-mcpu=cortex-m4 **-mfloat-abi=hard -mfpu=fpv4-sp-d16**")

LawranceLiu, 2021-05-27:
Welcome to reopen it if any further discussion.

closed 2021-05-27 by LawranceLiu