MediaTek-Labs/mt3620_m4_software #40: Problem with mbox communication

renakim, 2021-04-09:
We're developing and using an application including M4<->A7 communication based on the `MT3620_RTApp_BareMetal_MBOX` example. (M4<->M4 communication is not used)
- https://github.com/MediaTek-Labs/mt3620_m4_software/blob/master/MT3620_M4_Sample_Code/BareMetal/MT3620_RTApp_BareMetal_MBOX/RTAppA/main.c

However, an exceptional behavior occurred when communicating using mbox.
The situation is as follows.

- M4 -> A7 sending data
- received data in A7 -> parsing -> sending response to M4
- **No event occurs in M4**

The problem is that this doesn't happen every time, it happens randomly during execution.
When I repeated the test, the problem has also appeared after hundreds of runs.

When the mbox connection is established and data is sent from A7 through the `send()` function, is there a case where the transmission does not work? In this case, which part should I check?

LawranceLiu, 2021-04-13:
Hi Rena, 
Did you modify anything of the sample code? 
I tried but could not reproduce this issue. 

So, I tried to modify the sample code to be more aggressive. 
From:
    - HLApp sends the message to RTApps in every 3 seconds
To
    1. HLApp sends the message to RTApp_A & RTApp_B
    2. HLApp receives the message from  RTApp_A & RTApp_B, 
    3. repeat #1/#2 without any delay. 
After 1 day stress teat (more than 1 million test iterations), everything is still working fine.
Then, I have another guy using another board to do the same test, everything is sitll working fine.

Not sure if you modify anything of the sample code or do you have another board to do the same test?

LawranceLiu, 2021-05-27:
Welcome to reopen it if any further discussion.

closed 2021-05-27 by LawranceLiu