MediaTek-Labs/mt3620_m4_software #43: FPU related build error on MT3620 RT Core FreeRTOS example

IvanLee-1, 2023-04-21:
When I build an FreeRTOS example on github(https://github.com/MediaTek-Labs/mt3620_m4_software/tree/master/MT3620_M4_Sample_Code/FreeRTOS/MT3620_RTApp_FreeRTOS_CPP) There is an build error as below. and it seems that it's related to the FreeRTOS/portable/port.c

`[build] FAILED: lib/MT3620_M4_Driver/MT3620_M4_BSP/FreeRTOS/CMakeFiles/MT3620_M4_FreeRTOS.dir/portable/port.c.obj 
[build] C:\PROGRA~2\ARMGNU~1\12DBAB~1.2RE\bin\AR19DD~1.EXE -DOSAI_ENABLE_DMA -DOSAI_FREERTOS -IC:/lab1/mt3620_m4_software/MT3620_M4_Sample_Code/FreeRTOS/MT3620_RTApp_FreeRTOS_CPP/FreeRTOS_RTcore_CPP -IC:/lab1/mt3620_m4_software/MT3620_M4_Sample_Code/FreeRTOS/MT3620_RTApp_FreeRTOS_CPP/PUBLIC -IC:/lab1/mt3620_m4_software/MT3620_M4_Sample_Code/FreeRTOS/MT3620_RTApp_FreeRTOS_CPP/. -IC:/lab1/mt3620_m4_software/MT3620_M4_BSP/FreeRTOS/./portable -IC:/lab1/mt3620_m4_software/MT3620_M4_BSP/FreeRTOS/./include -isystem "C:/Program Files (x86)/Arm GNU Toolchain arm-none-eabi/12.2 rel1/lib/gcc/arm-none-eabi/12.2.1/include" -isystem "C:/Program Files (x86)/Arm GNU Toolchain arm-none-eabi/12.2 rel1/arm-none-eabi/include" -mcpu=cortex-m4 -g -std=c11 -Wall -g2 -gdwarf-2 -O0 -MD -MT lib/MT3620_M4_Driver/MT3620_M4_BSP/FreeRTOS/CMakeFiles/MT3620_M4_FreeRTOS.dir/portable/port.c.obj -MF lib\MT3620_M4_Driver\MT3620_M4_BSP\FreeRTOS\CMakeFiles\MT3620_M4_FreeRTOS.dir\portable\port.c.obj.d -o lib/MT3620_M4_Driver/MT3620_M4_BSP/FreeRTOS/CMakeFiles/MT3620_M4_FreeRTOS.dir/portable/port.c.obj -c C:/lab1/mt3620_m4_software/MT3620_M4_BSP/FreeRTOS/portable/port.c
[build] C:\Users\ivan\AppData\Local\Temp\cc3R28Mf.s: Assembler messages:
[build] C:\Users\ivan\AppData\Local\Temp\cc3R28Mf.s:857: Error: selected FPU does not support instruction -- `vstmdbeq r0!,{s16-s31}'
[build] C:\Users\ivan\AppData\Local\Temp\cc3R28Mf.s:879: Error: selected FPU does not support instruction -- `vldmiaeq r0!,{s16-s31}'
[build] ninja: build stopped: subcommand failed.`

How can I fix this build error?
Thanks.

bawilless, 2023-04-21:
@IvanLee-1,

Here are some notes on this error from a while back.  You should find that the projects  build under Visual Studio, but will see the issue when building with VS Code.  

Note from a while back:

You can see my changes by going to my Forked Repo and my VsCodAndLinuxBuilds branch here:  [Avnet/mt3620_m4_software at VsCodAndLinuxBuilds (github.com)](https://github.com/Avnet/mt3620_m4_software/tree/VsCodAndLinuxBuilds).  Look at the [mt3620_m4_software](https://github.com/Avnet/mt3620_m4_software/tree/VsCodAndLinuxBuilds)/[MT3620_M4_Sample_Code](https://github.com/Avnet/mt3620_m4_software/tree/VsCodAndLinuxBuilds/MT3620_M4_Sample_Code)/[FreeRTOS](https://github.com/Avnet/mt3620_m4_software/tree/VsCodAndLinuxBuilds/MT3620_M4_Sample_Code/FreeRTOS)/MT3620_RTApp_FreeRTOS_UART example and you’ll see the following changes . . . 

- CMakeLists.txt
-- Added the line: set(CMAKE_C_FLAGS "-fno-common -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -Wall") to get the application to build/link
- .vscode Directory
-- VS Code looks here for the settings and launch settings.

IvanLee-1, 2023-04-24:
@bawilless ,

Thanks for your support.
I confirmed that the FPU-related build error is solved on my side with your changes.

- CMakeLists.txt
-- Added the line: set(CMAKE_C_FLAGS "-fno-common -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -Wall") to get the application to build/link

Thanks.